hannobraun/fornjot #1619: For the problem that the mac cannot be used in time after installation

Jzow, 2023-02-24:
tag: #1612 

I conducted the second installation experiment on the mac operating machine again. There are two problems.

**First question:**
The mac will prompt that "fj-app-x86_64-apple-arwin" cannot be opened because it is from an unknown developer.

**Second question:**
Failed to open the document "fj app". Text encoding Unicode (UTF-8) is not applicable.

There are two situations here. The first is ordinary users and technicians. For technicians, they can accurately verify app security and grant permissions. This is fine, but ordinary users cannot solve the problem in a short time. Need refer to relevant information.

I checked the ci/cd process of github, which is only responsible for distributing binary files to different operating systems, but can't do some permission verification. These problems are all from the computer, **so I personally think that supplementary careful documents is the preferred solution at this stage.**

hannobraun, 2023-02-27:
Hey @Jzow, sorry for the late reply! I had other things on my mind over the weekend :smile: 

As I said elsewhere, I don't have access to macOS and know little about it, so I'm not sure what the best way to solve this is. Adding documentation on how to work around these issues sounds reasonable though. I think [the *Installation* section in `README.md`](https://github.com/hannobraun/Fornjot/blob/main/README.md#installation) is probably the best place for that, for the time being.

Jzow, 2023-02-27:
@hannobraun Ok, let me imagine how to supplement this document

hannobraun, 2023-05-15:
This issue is no longer applicable. The affected code [has been removed](https://github.com/hannobraun/fornjot/pull/1822). See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.

closed 2023-05-15 by hannobraun