hannobraun/fornjot #1935: Validate face orientation of `Solid`s

hannobraun, 2023-07-11:
The orientation of a face is defined by the winding of its exterior cycle. On the front side of the face, the exterior cycle is wound counter-clockwise. It is possible to create `Solid`s where some or all of the faces face inwards, which would lead to invalid shading when displaying the `Solid`, and possibly invalid meshes when exporting it to an external file format.

This should be caught in a validation check, which makes sure that all faces of every solid point outwards.

A-Walrus, 2023-07-21:
Some questions:

- Should we be checking that all faces of a `Shell` have the same orientation (which can be either inwards or outwards)?
- If a `Solid` has a cavity, what should be the orientation of the faces of the cavity?

hannobraun, 2023-07-22:
> Should we be checking that all faces of a `Shell` have the same orientation (which can be either inwards or outwards)?

Yeah, that makes sense! A mixed-orientation `Shell` would definitely be invalid (even though that's not specified anywhere right now), and having that check in place would then make the `Solid` check easier

> If a `Solid` has a cavity, what should be the orientation of the faces of the cavity?

The cavity is "outside" of the `Solid`. So if you're sitting inside of the cavity looking at the solid, you should see counter-clockwise faces.

hannobraun, 2023-11-28:
I've been looking into coordinate systems and cycle winding while working on https://github.com/hannobraun/fornjot/issues/2098, and I found out that the situation is not as simple as I presented it in the issue description above. I said that the orientation of a face is defined by the winding of its exterior cycle, and that is simply wrong. The orientation of a face is defined by the *combination of* the winding and the handedness of the surface coordinate system.

To summarize:

- When looking at a face **from the front** (i.e. looking at a shell from the outside):
  - Winding must be **counter-clockwise** and surface coordinate system must be **right-handed**, or
  - winding must be **clockwise** and surface coordinate system must be **left-handed**.
- When looking at a face **from the back** (i.e. looking at a shell from inside the shell):
  - Winding must be **counter-clockwise** and surface coordinate system must be **left-handed**, or
  - winding must be **clockwise** and surface coordinate system must be **right-handed**.

All of that is highly confusing, and there is [way too much](https://github.com/hannobraun/fornjot/blob/da350a6fce90cdc2475e3eddbd94d8fda74b136d/crates/fj-core/src/objects/kinds/cycle.rs#L27-L81) [code that must](https://github.com/hannobraun/fornjot/blob/da350a6fce90cdc2475e3eddbd94d8fda74b136d/crates/fj-core/src/objects/kinds/face.rs#L64-L69) [deal with this](https://github.com/hannobraun/fornjot/blob/da350a6fce90cdc2475e3eddbd94d8fda74b136d/crates/fj-core/src/algorithms/approx/face.rs#L121-L122) [in too many places](https://github.com/hannobraun/fornjot/blob/da350a6fce90cdc2475e3eddbd94d8fda74b136d/crates/fj-core/src/algorithms/triangulate/delaunay.rs#L58-L67).

I *hope* that it might be reasonable to just require faces to just *always be winded counter-clockwise and have a right-handed coordinate system* (when looking from the front), but I don't know. It would require reversing the surface when reversing a face, and if you do that, then you have coincident faces that aren't on the same surface (because I've reversed it; it's a different surface now). Which would break all kinds of assumptions that the code currently makes, or might make in the future.

So yeah, I'm confused. I'll keep thinking about it, but for now, if anyone wants to address this issue, the overview above should at least provide the correct set of rules to validate the orientation of faces.

felix91gr, 2023-11-28:
I hope these two cents are of a currency that works in this space. I hope they do >.<

If I recall correctly from class, the way face orientation is often coded for computer graphics purposes is by:
- First setting in stone a handedness for the coordinate system (there is one handedness that's used basically everywhere, but I forget which one of the two it is),
- Then, encoding the face orientation through the winding order of its vertices. If the winding coincides with the handedness of the coordinate system, the face is pointing outwards. If it isn't, the face is pointing inwards.

Basically what's done is that one of the two degrees of freedom is collapsed, and the other one is used to set the orientation.

---

Aside: there are other ways to orient faces, but the ones I know of require to add normal vectors to each vertex. To my eyes though, doing so seems unnecessarily wasteful in space for CAD, as well as numerically slightly less stable, than encoding face orientation in the winding order.

hannobraun, 2023-11-28:
> I hope these two cents are of a currency that works in this space >.<

I'll take them :smile: 

> If I recall correctly from class, the way face orientation is often coded for computer graphics purposes is by:
>
> [...]

Yeah, that's what I would like to do. It's what I meant above, where I wrote "*I hope that it might be reasonable...*".

To expand/clarify that, I know that this is a common approach in general, but there are problems with it within the context of Fornjot that would need to be addressed. Specifically, the way surfaces are handled.

The short version is,  Fornjot tries to make geometric relationships explicit. Two faces aren't just allowed to be coincident. If they are, they must refer to the same surface[^1]. This is done to handle numerical inaccuracy, and there's some [not-quite-up-to-date documentation about it](https://github.com/hannobraun/fornjot/blob/83691f6d6c6cd6d6f6a07fa1ba80ceab2d0937a3/crates/fj-core/src/lib.rs#L11-L80).

So, the problem is, if I have two coincident faces that face in different directions, they must per this rule reference the same surface. But since the coordinate system is defined by the surface, one of the faces must, necessarily, have a left-handed coordinate system.

I'm not sure what to do about that. Maybe the rule about surfaces is not exactly necessary, and we can do without it. Maybe a surface can have two coordinate systems (so each side gets a right-handed one). Or maybe coordinate systems can be decoupled from surfaces completely. Right now, it's not clear to me what the solution should be.

> Aside: there are other ways to orient faces, but the ones I know of require to add normal vectors to each vertex. To my eyes though, doing so seems unnecessarily wasteful in space for CAD, as well as numerically slightly less stable, than encoding face orientation in the winding order.

Yeah, I'd like to avoid that. I'd actually like to experiment with decoupling geometric and topological data from each other (which is a topic for another day; I'll open an issue about that when I can). Using normal vectors to define orientation would run counter to that.

[^1]: Actually, as of now, they need not. At least I don't think it's a documented requirement, and it's certainly not enforced. But that's more of an artifact of the current implementation. Half-edges and curves work exactly like that (i.e. coincident half-edges must refer to the same curve), and that's works very well.

hannobraun, 2025-06-23:
The validation code will most likely be replaced when the [experimental architecture](https://github.com/hannobraun/fornjot/tree/main/experiments/2025-03-18) is merged back into the mainline code. This issue may still apply then (the experimental code has no validation code to speak of, so far), but I think it would be a waste of time to fix this before then.

Labeling as "blocked" until the experimental code is merged.