hannobraun/fornjot #2011: Cannot display example models

bluthej, 2023-09-07:
## Problem

Tried displaying the example models to get started with Fornjot but I could not get any to run correctly.
<details><summary>Error</summary>
<p>

```bash
❯ cargo run -p cuboid
    Finished dev [unoptimized + debuginfo] target(s) in 0.11s
     Running `target/debug/cuboid`
warning: queue 0x56192adb08d0 destroyed while proxies still attached:
  xdg_wm_base@23 still attached
  wl_output@15 still attached
  wp_fractional_scale_manager_v1@14 still attached
  xdg_activation_v1@13 still attached
  zwp_text_input_manager_v3@12 still attached
  zwp_relative_pointer_manager_v1@11 still attached
  zwp_pointer_constraints_v1@10 still attached
  zxdg_decoration_manager_v1@9 still attached
  wl_seat@8 still attached
  wp_viewporter@7 still attached
  wl_subcompositor@6 still attached
  wl_compositor@5 still attached
  wl_shm@4 still attached
  wl_registry@2 still attached
Error: Display(GraphicsInit(RequestAdapter))
```

</p>
</details> 

Note that exporting the example models does work.

## Configuration

- OS: Arch Linux x86_64
- Kernel: 6.4.12-arch1-1
- CPU: AMD Ryzen 7 4800H with Radeon Graphics (16) @ 2
- DE: Hyprland

## Version

Cloned the main branch on the 7th of september 2023.

hannobraun, 2023-09-07:
Thank you, @bluthej! I'll have a look later.

hannobraun, 2023-09-07:
Hey @bluthej, I've submitted a number of pull requests that might or might not address this. Could you please run `RUST_LOG=fj_viewer=debug cargo run -p cuboid` and post the output?

bluthej, 2023-09-07:
@hannobraun I confirm that these changes fixed the issue :)  Thanks a lot for taking the time to investigate that!

I do get an error but it doesn't prevent the model from being displayed, so I guess this must be a minor thing. Here is the error in case you are interested:

```bash
2023-09-07T22:12:53.671118Z ERROR sctk_adwaita::config: XDG Settings Portal did not return response in time: timeout: 100ms, key: color-scheme    
```

hannobraun, 2023-09-08:
Nice! Could you still post the other log output? There should be a message about an adapter being selected. My theory is, that Vulkan is not available on your system for some reason, and OpenGL was selected instead (previously our settings would have prevented that, which is one of the things I changed). But I'd like to confirm.

> I do get an error but it doesn't prevent the model from being displayed

No idea what that is. It's not something Fornjot is using directly, so it must originate somewhere in the dependency chain. If it doesn't affect anything, I'd be inclined to ignore it and hope it goes away :joy: 

---

For the record, here's what I did: As I said before, our previous settings only allowed for wgpu's "primary" backends to be selected, which doesn't include OpenGL. I changed it so all backends are acceptable, and also fixed an OpenGL-only bug that I ran into on my system.

All those "still attached" warnings seem to just be a consequence of the aborted initialization. If I return the `RequestAdapter` error manually, I'm seeing the same warnings, even if an adapter was actually successfully selected.

closed 2023-09-08 by hannobraun

bluthej, 2023-09-08:
This is all I get in my terminal when I run  the command:

```bash
❯ cargo run -p cuboid
    Finished dev [unoptimized + debuginfo] target(s) in 0.52s
     Running `target/debug/cuboid`
2023-09-08T10:00:16.483482Z ERROR sctk_adwaita::config: XDG Settings Portal did not return response in time: timeout: 100ms, key: color-scheme
```

I don't have any other message printed out. If there is something I'm missing could you tell me where to look for a more complete log? I tried the verbose option of the CLI but that only added some cargo output.

hannobraun, 2023-09-08:
To see the debug output, you need the full command I quoted above:
```
RUST_LOG=fj_viewer=debug cargo run -p cuboid
```

bluthej, 2023-09-08:
Oh yes sorry I totally overlooked your instructions... Here is the debug output:

```bash
❯ RUST_LOG=fj_viewer=debug cargo run -p cuboid
    Finished dev [unoptimized + debuginfo] target(s) in 0.53s
     Running `target/debug/cuboid`
2023-09-08T17:00:05.742866Z DEBUG fj_viewer::graphics::renderer: Using adapter: AdapterInfo { name: "AMD Radeon Graphics (renoir, LLVM 16.0.6, DRM 3.52, 6.4.12-arch1-1)", vendor: 4098, device: 0, device_type: Other, driver: "", driver_info: "", backend: Gl }

```

hannobraun, 2023-09-08:
Thanks, that confirms my suspicion! For whatever reason, Vulkan is not available on your machine (at least to wgpu).