hannobraun/fornjot #2034: Cannot display example models due to "Null hardware" error

riskable, 2023-09-27:
When trying to run the cuboid example it crashes with the following output:

```
fornjot on  main via 🦀 v1.72.1 took 18s 
➜ RUST_LOG=fj_viewer=debug cargo run -p cuboid
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
     Running `target/debug/cuboid`
2023-09-27T14:22:32.517366Z DEBUG fj_viewer::graphics::renderer: Using adapter: AdapterInfo { name: "Null hardware (RADV NAVI10)", vendor: 4098, device: 29456, device_type: DiscreteGpu, driver: "radv", driver_info: "Mesa 23.0.4-0ubuntu1~23.04.1", backend: Vulkan }
Error: Error displaying model

Caused by:
    0: Error initializing graphics
    1: Error requesting device: Requesting a device failed
    2: Requesting a device failed
```

I suspect that this is caused by fornjot not knowing which GPU to use when it runs and is simply choosing the first card it "sees" (however it enumerates GPUs if it even does that).  My system has an AMD 7950 CPU which has a built-in GPU and an AMD Radeon RX 6650 XT discreet GPU.  In `/dev/dri` I get two cards:

```
➜ ls -l /dev/dri
total 0
drwxr-xr-x  2 root root        120 Sep 25 13:42 by-path
crw-rw----+ 1 root video  226,   0 Sep 25 13:42 card0
crw-rw----+ 1 root video  226,   1 Sep 25 13:42 card1
crw-rw----+ 1 root render 226, 128 Sep 25 13:42 renderD128
crw-rw----+ 1 root render 226, 129 Sep 25 13:42 renderD129
```

In this case, `card0`/`renderD128` is the built-in (unused) GPU that's not hooked up to any displays and `card1/`render129` is the discreet Radeon RX 6650 XT card that's connected to my display/running my desktop.

Not sure how to fix it (if that's the problem) but any mechanism that discards unused GPUs (for display... Not for the actual math work) or provides a mechanism to select which GPU to use would probably be fine.

hannobraun, 2023-09-27:
Thank you for opening this issue, @riskable! I've just merged [a pull request](https://github.com/hannobraun/fornjot/pull/2035) that adds some additional logging. I'd appreciate if you could run the same command again (`RUST_LOG=fj_viewer=debug cargo run -p cuboid`) and post the output.

riskable, 2023-09-28:
It still crashes but we get some more log details:

```
fornjot on  main via 🦀 v1.72.1 
➜ RUST_LOG=fj_viewer=debug cargo run -p cuboid
   Compiling fj-viewer v0.47.0 (/home/riskable/downloads/fornjot/crates/fj-viewer)
   Compiling fj-core v0.47.0 (/home/riskable/downloads/fornjot/crates/fj-core)
   Compiling fj-window v0.47.0 (/home/riskable/downloads/fornjot/crates/fj-window)
   Compiling fj v0.47.0 (/home/riskable/downloads/fornjot/crates/fj)
   Compiling cuboid v0.1.0 (/home/riskable/downloads/fornjot/models/cuboid)
    Finished dev [unoptimized + debuginfo] target(s) in 5.03s
     Running `target/debug/cuboid`
2023-09-28T18:56:51.677112Z DEBUG fj_viewer::graphics::renderer: Available adapter: AdapterInfo { name: "Null hardware (RADV NAVI10)", vendor: 4098, device: 29456, device_type: DiscreteGpu, driver: "radv", driver_info: "Mesa 23.0.4-0ubuntu1~23.04.1", backend: Vulkan }
2023-09-28T18:56:51.677161Z DEBUG fj_viewer::graphics::renderer: Available adapter: AdapterInfo { name: "llvmpipe (LLVM 15.0.7, 256 bits)", vendor: 65541, device: 0, device_type: Cpu, driver: "llvmpipe", driver_info: "Mesa 23.0.4-0ubuntu1~23.04.1 (LLVM 15.0.7)", backend: Vulkan }
2023-09-28T18:56:51.677171Z DEBUG fj_viewer::graphics::renderer: Available adapter: AdapterInfo { name: "NAVI23 (navi23, LLVM 15.0.3, DRM 3.49, 6.2.0-20-generic)", vendor: 4098, device: 0, device_type: Other, driver: "", driver_info: "", backend: Gl }
2023-09-28T18:56:51.680362Z DEBUG fj_viewer::graphics::renderer: Using adapter: AdapterInfo { name: "Null hardware (RADV NAVI10)", vendor: 4098, device: 29456, device_type: DiscreteGpu, driver: "radv", driver_info: "Mesa 23.0.4-0ubuntu1~23.04.1", backend: Vulkan }
Error: Error displaying model

Caused by:
    0: Error initializing graphics
    1: Error requesting device
    2: Requesting a device failed
```

riskable, 2023-09-28:
FYI: Just did a `cargo clean; cargo update` and re-ran the same command.  It still crashes with the same output :shrug:

hannobraun, 2023-09-29:
It's not surprising that it still crashes. The pull request just adds additional logging (plus a refactoring that might come in handy for a fix I'd like to attempt).

That additional log output is interesting, but doesn't make me hopeful. The "Null hardware" adapter that doesn't work is the only one that is associated with your discrete CPU. The second one is a CPU-based software renderer, and the third one is not looking like it'll work either (missing any information on the device and driver).

I'll see what I can do, but this might turn out to be a problem on the wgpu or driver level. If that's the case, I'm out of my depth.

hannobraun, 2023-09-29:
Could you give this another try, @riskable?

I've added code in https://github.com/hannobraun/fornjot/pull/2038 that tries all adapters until one of them works. I *hope* that will help in your case.

Regardless of whether it works or not, I'd be very interested in seeing what your debug output looks like now.

riskable, 2023-09-29:
@hannobraun The latest update fixes it!  The cuboid example now displays properly 👍.

I can even spin it around and whatnot with my mouse. Perfect.

closed 2023-09-29 by riskable