hannobraun/fornjot #2145: Consider adding Liberapay as donation option

hkienle, 2023-12-17:
https://liberapay.com

hannobraun, 2023-12-17:
Thank you for the suggestion, @hkienle!

I used to have a Liberapay profile, but decided to close it. It was additional work (another place where I had to maintain a profile, and on the bookkeeping side), and at the time I closed it, I had no more sponsors there, and hadn't for a while. It did not seem worth the trouble to maintain it.

Would you like to sponsor me on Liberapay? If so, I'd be happy to reopen my account and keep it open for as long as I'm being sponsored there.

hannobraun, 2024-01-12:
Closing due to lack of feedback.

If anybody would like to sponsor me, but can't or won't use GitHub Sponsors, please always feel free to contact me! I'm happy to accommodate you.

But at this point, I won't proactively maintain profiles in more places.

closed 2024-01-12 by hannobraun