hannobraun/fornjot #2307: Request: Add variant of `process_model` that accepts `Args` rather than parsing them

starthal, 2024-03-28:
Fornjot already provides a `process_model` helper that accepts a "standard" set of arguments for exporting or viewing.

I want to make an app that uses arguments to configure/parameterize a model. But `process_model` calls `parse` directly so I can't extend the arguments with my own.

With an API like `fn process_model_with_args<M>(fj: &mut fj::Instance, model: &M, args: fj::Args) -> fj::Result` this would be straightforward.

For example:

```rust
#[derive(clap::Parser)]
pub struct Parameters {
    #[clap(required = true, long, value_parser = parse_vector_3)]
    /// Size of the cuboid, as a comma-separated vector `x,y,z`
    size: [f64; 3],
    #[command(flatten)]
    /// Standard fj args
    fj: fj::Args,
}

fn parse_vector_3(arg: &str) -> anyhow::Result<[f64; 3]> {
    Ok(arg
        .split(',')
        .map(str::parse)
        .collect::<Result<Vec<f64>, _>>()?
        .as_slice()
        .try_into()?)
}

fn main() -> anyhow::Result<()> {
    let mut fj = fj::Instance::new();
    let p = <Parameters as clap::Parser>::parse();
    let model = cuboid::model(p.size, &mut fj.core);
    process_model_with_args(&mut fj, &model, p.fj)?;
    Ok(())
}
```

The user can invoke this like:

```
cargo run -- --size 3,5,7 --export cuboid.stl
```

hannobraun, 2024-03-30:
Thank you for opening this issue, @starthal! That's a very reasonable request.

Happy to accept a pull request implementing what you propose. Otherwise, I'll get this done whenever I can get to it.

hannobraun, 2024-04-03:
Labeling as `good first issue`, as this is a pretty small change, and the solution is already laid out in the issue description.

hannobraun, 2025-05-26:
Addressed in https://github.com/hannobraun/fornjot/pull/2527.

closed 2025-05-26 by hannobraun