hannobraun/fornjot #2437: Regression: Holes model example exits with panic

pabu, 2024-08-03:
The holes model example exits with panic since this commit that changes how to construct a circle:

https://github.com/hannobraun/fornjot/commit/39b0274a92f119a763196cb38a74367cc0377639

Looks like it panics trying to create an approximation of the solid. It is very clear what the commit changes so presumably this is also a quick task for someone familiar with the codebase.

Perhaps to prevent similar things in the future, the demo models could be tested in a CI pipeline but that is out of scope for this ticket.

Sergi-Ferrez, 2024-08-03:
What OS are you using?
I think this has relation to #2376

hannobraun, 2024-08-05:
Thank you for opening this issue, @pabu! Sorry for the late reply.

This indeed looks like a duplicate of https://github.com/hannobraun/fornjot/issues/2376 (thanks for linking that, @Sergi-Ferrez). The commit you linked is from the pull request where that problem first appeared (https://github.com/hannobraun/fornjot/pull/2375).

> Looks like it panics trying to create an approximation of the solid. It is very clear what the commit changes so presumably this is also a quick task for someone familiar with the codebase.

Unfortunately it's not easy to debug, as it's totally unclear why that problem would appear _just on Windows_. What this needs right now is someone with access to Windows to debug it. I'd expect that to be hard, as these kinds of issues tend to be hard to debug for me too.

If nobody else can fix it, I'm hoping that my ongoing work on the geometry code will fix this as a side effect, at some point (which seems not unlikely). If that doesn't help either, I'll probably have to bite the bullet and try to debug this via the CI server. But that's going to be a huge pain, and for the time being, I have other priorities.

> Perhaps to prevent similar things in the future, the demo models could be tested in a CI pipeline but that is out of scope for this ticket.

That is already happening. Just not on Windows, as per https://github.com/hannobraun/fornjot/issues/2376.

---

@pabu, can you please confirm what platform you're running on? If it's Windows, we can close this as a duplicate.