hannobraun/fornjot #2507: Feature Request: Support for hex color codes in Color

MilliUser, 2025-03-16:
## Feature Request: Add Hexadecimal Color Support to the `Color` Struct

### Problem
Hexadecimal color codes (`#RRGGBB` or `#RRGGBBAA`) are widely used in web development and design tools. Currently, the `Color` struct does not support creating colors directly from hex strings, which limits its usability in some scenarios.

### Proposed Solution
I propose adding a `From<&str>` implementation to the `Color` struct to allow conversion from hex color codes. This would enable users to easily create colors from hex strings like `#FF5733` or `#FF573380`.

### Example Implementation
Here is how the implementation could look:
crates/fj-interop/src/color.rs
```rust
...
impl From<&str> for Color {
    fn from(hex: &str) -> Self {
        let hex = hex.trim_start_matches('#');
        let len = hex.len();

        let parse_hex = |start: usize, end: usize| -> u8 {
            u8::from_str_radix(&hex[start..end], 16).unwrap_or(0)
        };

        match len {
            6 => {
                let r = parse_hex(0, 2);
                let g = parse_hex(2, 4);
                let b = parse_hex(4, 6);
                Self([r, g, b, 255])
            }
            8 => {
                let r = parse_hex(0, 2);
                let g = parse_hex(2, 4);
                let b = parse_hex(4, 6);
                let a = parse_hex(6, 8);
                Self([r, g, b, a])
            }
            _ => Self::default(),
        }
    }
}
```
### Benefits
This feature would simplify color creation from hex codes, which are commonly used in web and graphic design. It would also improve interoperability with other tools and libraries that use hex colors.

hannobraun, 2025-03-17:
Thank you for opening this issue, @MilliUser. This request makes sense. I am not going to work on this though, as I'm still focused on the ongoing [series of experiments](https://github.com/hannobraun/fornjot/tree/main/experiments).

In principle, I'd be happy to accept a pull request in the meantime, but that may be wasted effort: At this point, it's unclear how much of the old code will survive the eventual merge of the experimental code back into the mainline. And in fact, I've been considering going the opposite direction with this: Declare color (and anything more advanced like textures) as out of scope for this library. This is something that can be built on top.

MilliUser, 2025-03-17:
Thank you for your response and for considering my request! I appreciate the transparency about the current focus on experiments and the potential future direction of the library. I completely understand that the codebase is undergoing significant changes, and I respect the decision to potentially keep features like color and textures out of scope for the core library.

I’ll keep an eye on the project’s progress and look forward to seeing where it goes in the future. If the opportunity arises, I’d be happy to contribute or explore building additional layers on top of the library for features like color support.

Thanks again for your time and insights!

hannobraun, 2025-03-21:
I have an update on what I wrote before. I said:

> At this point, it's unclear how much of the old code will survive the eventual merge of the experimental code back into the mainline.

While this is still broadly true, I've just finished deleting the experimental graphics code. The latest experiment is now using the regular mainline graphics code, and I'm pretty sure that is safe from replacement going forward.

This means that the improvement suggested here can be implemented any time, and any such effort is unlikely to be wasted. So if somebody wants to go ahead and file a pull request, you're welcome to!

Regarding this:

> And in fact, I've been considering going the opposite direction with this: Declare color (and anything more advanced like textures) as out of scope for this library. This is something that can be built on top.

I'm pretty sure I won't merge anything like textures any time soon, but simple colors will keep being support for debug purposes.

hannobraun, 2025-05-20:
Implemented in https://github.com/hannobraun/fornjot/pull/2526.

closed 2025-05-20 by hannobraun