hannobraun/fornjot #2508: Suggestion: Add a list of notable projects that use this library to the README

DeflateAwning, 2025-03-16:
Are there any projects that use this library yet? It might be nice to track them as a list in the README to be able to see how much it gets used

hannobraun, 2025-03-17:
Thank you for opening this issue, @DeflateAwning, that's a good point! I actually had a task somewhere, but I never got around to it and now I can no longer find it.

Anyway, here's the list of projects that I'm currently aware of:

- [bevy_mod_fornjot](https://github.com/TotalKrill/bevy_mod_fornjot)
- [CommandCAD](https://github.com/IamTheCarl/CommandCAD)
- [Fornjot Text](https://github.com/samgoldman/fj-text)

I don't think any of them are active right now.

hannobraun, 2026-02-25:
I actually added the projects I mentioned here to the README a while ago: https://github.com/hannobraun/fornjot?tab=readme-ov-file#status

Closing.